sparkmicro/Ki-nTree #259: Confusion re: content and use of kicad settings "Symbol Templates Folder"

plocher, 2024-10-08:
I am at a loss as to what should be in this field.

The settings Kicad  Symbol Libraries Folder are set to
/Volumes/Macintosh HD/Users/jplocher/Dropbox/KiCad/Libraries/Symbols

Whatever settings I try to use for  Kicad Symbol Templates Folder, when adding a part,  I get

Symbol Library: [SPCoast ]
Symbol Template: [Capacitors/Ceramic]
Footprint Library: [SPCoast]
Footprint[C_0603_1608Metric]

[KCAD]	Error loading template file (/Volumes/Macintosh HD/Users/jplocher/Dropbox/KiCad/Libraries/Symbolscapacitor.kicad_sym)

Needless to say, there is no file of that name, nor is there a symbol called "capacitor"...

Searches for kintree docs on "symbol templates folder" come up empty...

Where does "capacitor" come from, and what should I use for the templates folder?

Other than this hiccup, I'm impressed!

plocher, 2024-10-09:
More detail:
If I create an empty symbol library called capacitor.kicad_sym, I get an erronious error message:

[KCAD]	Error: Found more than 1 symbol template in template file, aborting

[MAIN]	Creating parameters
[TREE]	Error: Parameter template "Symbol" does not exist
[TREE]	Warning: The following parameters were skipped:
--->	Symbol

plocher, 2024-10-09:
If I place a symbol into the formerly empty capacitor.kicad_sym library named "C", I get
[KCAD]	Warning: Component C already in library

[MAIN]	Creating parameters
[TREE]	Error: Parameter template "Symbol" does not exist
[TREE]	Warning: The following parameters were skipped:
--->	Symbol

plocher, 2024-10-09:
and if I rename the "C" to "Symbol", I get

[MAIN]	Adding part to KiCad
[KCAD]	library_path: /Volumes/Macintosh HD/Users/jplocher/Dropbox/KiCad/Libraries/Symbols/SPCoast.kicad_sym
[KCAD]	Number of parts in library SPCoast.kicad_sym: 6
[KCAD]	Success: Component added to library SPCoast.kicad_sym

[MAIN]	Creating parameters
[TREE]	Error: Parameter template "Symbol" does not exist
[TREE]	Warning: The following parameters were skipped:
--->	Symbol

plocher, 2024-10-09:
It appears that:

A kicad_sym file with a single symbol named "Symbol" needs to exist in the Kicad Symbol Templates folder, with an arbitrary name that does not match either the IPN category code ("CAP") or the Inventree Category ("Components/Capacitor")

I'm guessing it is a lower cased version of the last segment of the Inventree Category.

Can you point me to where this is documented?  :-)

plocher, 2024-10-09:
Finally, where is the error

[TREE]	Error: Parameter template "Symbol" does not exist
[TREE]	Warning: The following parameters were skipped:
--->	Symbol

coming from?

T0jan, 2024-10-11:
> Needless to say, there is no file of that name, nor is there a symbol called "capacitor"...
>
> Searches for kintree docs on "symbol templates folder" come up empty...
>
> Where does "capacitor" come from, and what should I use for the templates folder?

Ki-nTree comes with it's own set of templates which will be mapped by default with config files, that's where the `capacitors.kicad_sym` comes from. You can create own symbol template files and put them wherever you want but the defaults probably give a good introduction and should cover many cases already. The most important thing is that a template symbol file can only contain one symbol (that's where your second error comes from).

In general the template files should be in your Ki-nTree installation path (usually in your python library): `{Ki-nTree PATH}/kicad/templates`

> [TREE] Error: Parameter template "Symbol" does not exist
> [TREE] Warning: The following parameters were skipped:
> ---> Symbol

That's a Ki-nTree InvenTree interaction error. Ki-nTree adds a symbol parameter to the newly created InvenTree part, to have the information about the KiCad library and symbol name also in InvenTree. Therefore a part parameter template named `Symbol` and also one for `Footprint` need to be set up in InvenTree. 

Have you run the setup_inventree mentioned in the [README](https://github.com/sparkmicro/Ki-nTree#before-starting)?