sparkmicro/Ki-nTree #260: Enhancements: Suppliers "not in use" behave unexpectidly

plocher, 2024-10-09:
I used settings to disable the suppliers I don't use / have API keys for, and ...

The "not In-Use" setting doesn't remove them from the dropdown list of suppliers used on the search page (IMO, the list should be pruned...), and

When only one supplier is still in-use, it could be auto-selected rather than leaving the field empty and forcing the user to pull down to select the only enabled supplier.

T0jan, 2024-10-11:
The drop-down list should update and remove disabled suppliers as soon as you save the change in the settings, please describe your full string of actions where you expect the list to be updated but it doesn't do it to see if we have a bug here.

selecting the single supplier automatically is a good idea we could probably soonish.

eeintech, 2024-10-16:
@plocher Don't forget to click on "Save" after you disable a supplier:

[image]

It does remove the supplier from the list, using the latest release:

[image]