sparkmicro/Ki-nTree #265: Mouser Api doesn't work anymore?

simoneamadori, 2024-10-25:
In the last few days I cannot do the mouser search anymore, going to the api part and testing tells :
[image]

I've recreated the api, but the problem persists.
Any idea where I can check/control? 

They updated the API?

simoneamadori, 2024-10-25:
Update:
if a do a search with the mouser part number I can load the data: (80-C0603C104K5RAC)
[image]
If I search with the part number C0603C104K5RAC it doesn't find it
[image]

simoneamadori, 2024-10-25:
In version 1.1.3 the mouser search works correctly search and test under supplier

simoneamadori, 2024-10-25:
in the terminal jsut:
```
←[95m
[MAIN]  Mouser search for C0603C104K5RAC←[0m
←[91m[INFO]     Error: Failed to fetch data for "C0603C104K5RAC"←[0m
←[95m
```
I've tried also with the test code, in this case the error is also with the mouser code:
```
←[95m
[MAIN]  Mouser search for DMP2066LSN-7←[0m
←[91m[INFO]     Error: Failed to fetch data for "DMP2066LSN-7"←[0m
←[95m
[MAIN]  Mouser search for 621-DMP2066LSN-7←[0m
←[91m[INFO]     Error: Failed to fetch data for "621-DMP2066LSN-7"←[0m
```
Also tried the uninstall/install but the problem remains, there's same cache to clean up?

T0jan, 2024-10-25:
does the connection test in the settings --> Supplier --> Mouser work?
And do you get info for any mouser part number you enter or was `80-C0603C104K5RAC` maybe just cached data?

simoneamadori, 2024-10-25:
Doesn't work, gives error.
probably is cached data, now on few tries it gives error:
```
←[95m
[MAIN]  Mouser search for 35605R6JT←[0m
←[91m[INFO]     Error: Failed to fetch data for "35605R6JT"←[0m
←[95m
[MAIN]  Mouser search for 279-35605R6JT←[0m
←[91m[INFO]     Error: Failed to fetch data for "279-35605R6JT"←[0m
←[95m
[MAIN]  Mouser search for VS-30BQ015-M3←[0m
←[91m[INFO]     Error: Failed to fetch data for "VS-30BQ015-M3"←[0m
←[95m
[MAIN]  Mouser search for 78-VS-30BQ015-M39AT←[0m
←[91m[INFO]     Error: Failed to fetch data for "78-VS-30BQ015-M39AT"←[0m
```
If it can help I'm in italy

T0jan, 2024-10-25:
Ok I have an idea why. 1.1.4 does not pull the newest version of the python API wrapper by default so you are probably still on the old version which can not work with 1.1.4.

to fix this run `pip install -U mouser==0.1.6` or the corresponding command to install the needed version manually. this is already fixed for the next version in the `requirements.txt`

simoneamadori, 2024-10-25:
I've updated from 0.1.5 to 0.1.6 and that fixed it, thanks!

closed 2024-10-25 by simoneamadori

eeintech, 2024-10-25:
Thanks @T0jan for the quick resolution!