sparkmicro/Ki-nTree #266: Deafult Templates path not accessible

Keybored02, 2024-10-25:
Hi
Couple of info first: latest kintree version, installed via pip, on a Win10 machine.

The default template path seems not directly accessible via the GUI. This means that no templates appear in the drop down menu. When trying to modify it via the settings, the browse button doesn't open a file explorer, it just briefly blinks. Modifying the source file `kicad.yaml` in the package directory or the cached file in the user directory to point to the full path of the template dir fixes the issue. Seems like the default path `kintree/kicad/templates/` might not be recognized as a valid. 

Let me know if you need further info to test whether it's a one of kind situation. Reinstalling does not affect it.

eeintech, 2024-10-25:
Hello @Keybored02 

I reproduced your issue and, the Ki-nTree Settings view, the browsing feature seems to be broken with the default path (eg. `kintree/kicad/templates/`) present in the field:
[image]

But if you delete the entry then it works and you can pick the path to the templates:
[image]

> Note that you'll have to download the templates from this repo: https://github.com/sparkmicro/Ki-nTree/tree/main/kintree/kicad/templates

Then, back to the normal view and the list of templates should load normally:
[image]

Hopefully that's a good workaround for you too, probably need to fix that broken Browse behavior! 😄

Keybored02, 2024-10-25:
Hi, thanks for the quick response! I could get it to work by modifying the .yaml file, so I'm definitely good for now. Great to know there's a quicker solution though.

closed 2024-12-29 by eeintech