sparkmicro/Ki-nTree #268: Inventree parameter not created if Part Category Parameter Template exists for parameter

ursincantieni, 2024-12-21:
Running on Ki-nTree version `1.1.4`.

I'm creating Inventree parts from digikey search with the following `supplier_parameters.yaml`
```
Capacitor:
  Tolerance:
  - Tolerance
  Value:
  - Capacitance
  Voltage:
  - Voltage - Rated
  - Voltage Rated
```
The issue I'm seeing is that upon creating a part some parameters are skipped, i.e. `Tolerance` and `Voltage`, while others work as intended, i.e. `Value`.  While I have set up Part Category Parameter Templates for `Tolerance` and `Voltage` in Inventree Admin tool I removed it for `Value`. 
```
[MAIN]	Creating parameters
[INFO]	Success: The following parameters were created:
--->	Value
[TREE]	Warning: The following parameters were skipped:
--->	Tolerance
--->	Voltage
```
After removing the Parameter template for `Voltage` as well, the parameter will be created and populated correctly.
```
[MAIN]	Creating parameters
[INFO]	Success: The following parameters were created:
--->	Value
--->	Voltage
[TREE]	Warning: The following parameters were skipped:
--->	Tolerance
```

It appears that the parameters are skipped if they are already created as an effect of the template. This is not an issue by itself, as they do not need to be created (again) because they already exist (due to the template). The issue, however, is that for the skipped parameters the value will not be populated and left empty.

A workaround is to update the create part after creation by checking the `Update existing` box in invenTree page. The fact that this work also tells me that the linking of the parameters generally is setup correctly. However, this is an unnecessary manual step that requires me to copy paste the IPN from InvenTree back into Ki-nTree and run the update.

```
[MAIN]	Creating parameters
[INFO]	Success: The following parameters were updated:
--->	Tolerance
[TREE]	Warning: The following parameters were skipped:
--->	Value
--->	Voltage
```

ursincantieni, 2024-12-21:
Turns out `settings.UPDATE_INVENTREE` is `False` here when creating a part, blocking the `was_updated` from being set to true.

https://github.com/sparkmicro/Ki-nTree/blob/345304ec49671de7e464588c1687f49ba1cfce2d/kintree/database/inventree_api.py#L866-L882

eeintech, 2024-12-29:
> A workaround is to update the create part after creation by checking the `Update existing` box in invenTree page. The fact that this work also tells me that the linking of the parameters generally [...]

This is designed this way to prevent overwriting data by default.

The first time you add the part, all parameters should be added and you shouldn't have to enable that option, if that's not the case either there is an issue with `supplier_parameters.yaml` or missing parameter templates.

ursincantieni, 2024-12-29:
> The first time you add the part, all parameters should be added and you shouldn't have to enable that option, if that's not the case either there is an issue with supplier_parameters.yaml or missing parameter templates.

I believe what happens is that when setting a `Part Category Parameter Template` in Inventree this Parameter is automatically created upon Part Creation. Example: I have a Category `Capacitor` and a Part Category Parameter Template for the Parameter `Value` in the Category `Capacitor`. If I now use Kintree to add a new Part to the `Capacitor` Category, Kintree will
1. create a new part in the `Capacitor` category. Because of the Part Category Parameter Template, Inventree will automatically add a parameter `Value` with a default value (or empty).
2. Kintree goes over all parameters in `supplier_parameters.yaml`, creating new Part Parameters for the ones that don't exist yet. It will skip `Value` because it was already created as an effect of the template.

As a user, I would expect this parameter to be updated with the right value in this case. Do you agree?

Perhaps skipping the `settings.UPDATE_INVENTREE` check is not an optimal solution. I think it shouldn't overwrite data that the user is unaware of, since it will only update parameters in the user configured `supplier_parameters.yaml`.

eeintech, 2024-12-29:
Ok I see what you mean now, thanks for clarifying. 

I had forgotten that Part Category templates existed, and yes if the value is empty it will stay empty...
It will have to be looked at more to make sure we handle those too.