sparkmicro/Ki-nTree #274: Testing endpoints results in a crash

FilipSzkandera, 2025-01-16:
After installing Kintree I've tried to connect to my Invetree instance, but the python crashed when I've pressed the "test" button. Same error when I've tried adding Mouser as a supplier (inserting my API key and pressing "save" -> "test").

Reinstallation did not resolve the issue.

The error is the same in both cases:

**Error:** 
```bash
objc[31921]: +[NSString initialize] may have been in progress in another thread when fork() was called.
objc[31921]: +[NSString initialize] may have been in progress in another thread when fork() was called. We cannot safely call it or ignore it in the fork() child process. Crashing instead. Set a breakpoint on objc_initializeAfterForkError to debug.
```

**Versions:**
Kintree 1.2.0
Python env 3.12

T0jan, 2025-01-16:
Please add more information, especially about your system, because with the parameters you listed we saw no such problems on various Linux and Windows versions.

Also please add the full terminal output of the Ki-nTree session and not just the two error lines.

FilipSzkandera, 2025-01-16:
@T0jan ofc, this can be Mac issue, I can try it on my linux machine later. 

System: Macos sequoia 15.0.1 (24A348)

But for the terminal session, there is nothing more; (This is after clicking "test" on Inventree instance. The supplier "test" has the same result for me, but the last error line is not there)

```bash
(env) user@user kintree-1 % kintree
objc[33413]: +[NSString initialize] may have been in progress in another thread when fork() was called.
objc[33413]: +[NSString initialize] may have been in progress in another thread when fork() was called. We cannot safely call it or ignore it in the fork() child process. Crashing instead. Set a breakpoint on objc_initializeAfterForkError to debug.
[TREE]	Error connecting to InvenTree server: invalid address, username or password
```
And there is a popup with "Python quit unexpectedly."

I've downloaded the latest 1.2.0 release, went into the downloaded folder and
```bash
pip3 install kintree-1.2.0-py3-none-any.whl
```

T0jan, 2025-01-16:
Hmm in theory it should work on MacOS but I have no way of testing it properly. It looks like it crashes when a pop-up is meant to appear. The error message makes it seem that there is a problem deep in one of our dependencies, most likely flet. So for now I would say there isn't really anything we can do to fix it sadly. 

There are some similar issues out there which seem to be worked around by setting an environmental variable:

```OBJC_DISABLE_INITIALIZE_FORK_SAFETY=YES```