sparkmicro/Ki-nTree #276: 'kintree' is not recognized as an internal or external command on Windows11

martin323232, 2025-02-22:
Hello, 

installed kintree using 'pip install -U kintree'

`Requirement already satisfied: kintree in c:\users\martinmaursundrishov\appdata\local\packages\pythonsoftwarefoundation.python.3.11_qbz5n2kfra8p0\localcache\local-packages\python311\site-packages (1.1.4)
Requirement already satisfied: cloudscraper<2.0.0,>=1.2.71 in c:\users\martinmaursundrishov\appdata\local\packages\pythonsoftwarefoundation.python.3.11_qbz5n2kfra8p0\localcache\local-packages\python311\site-packages (from kintree) (1.2.71)
Requirement already satisfied: digikey-api<2.0.0,>=1.0.0 in c:\users\martinmaursundrishov\appdata\local\packages\pythonsoftwarefoundation.python.3.11_qbz5n2kfra8p0\localcache\local-packages\python311\site-packages (from kintree) (1.1.0)
Requirement already satisfied: flet<0.25.0,>=0.24.1 in c:\users\martinmaursundrishov\appdata\local\packages\pythonsoftwarefoundation.python.3.11_qbz5n2kfra8p0\localcache\local-packages\python311\site-packages (from kintree) (0.24.1)
Requirement already satisfied: inventree<0.18.0,>=0.17.0 in c:\users\martinmaursundrishov\appdata\local\packages\pythonsoftwarefoundation.python.3.11_qbz5n2kfra8p0\localcache\local-packages\python311\site-packages (from kintree) (0.17.4)
Requirement already satisfied: kiutils<2.0.0,>=1.4.0 in c:\users\martinmaursundrishov\appdata\local\packages\pythonsoftwarefoundation.python.3.11_qbz5n2kfra8p0\localcache\local-packages\python311\site-packages (from kintree) (1.4.8)
Requirement already satisfied: mouser<0.2.0,>=0.1.3 in c:\users\martinmaursundrishov\appdata\local\packages\pythonsoftwarefoundation.python.3.11_qbz5n2kfra8p0\localcache\local-packages\python311\site-packages (from kintree) (0.1.6)
Requirement already satisfied: multiprocess<0.71.0,>=0.70.12 in c:\users\martinmaursundrishov\appdata\local\packages\pythonsoftwarefoundation.python.3.11_qbz5n2kfra8p0\localcache\local-packages\python311\site-packages (from kintree) (0.70.17)
Requirement already satisfied: pyyaml<7.0.0,>=6.0.1 in c:\users\martinmaursundrishov\appdata\local\packages\pythonsoftwarefoundation.python.3.11_qbz5n2kfra8p0\localcache\local-packages\python311\site-packages (from kintree) (6.0.2)
Requirement already satisfied: thefuzz<0.20.0,>=0.19.0 in c:\users\martinmaursundrishov\appdata\local\packages\pythonsoftwarefoundation.python.3.11_qbz5n2kfra8p0\localcache\local-packages\python311\site-packages (from kintree) (0.19.0)
Requirement already satisfied: validators<0.20.0,>=0.19.0 in c:\users\martinmaursundrishov\appdata\local\packages\pythonsoftwarefoundation.python.3.11_qbz5n2kfra8p0\localcache\local-packages\python311\site-packages (from kintree) (0.19.0)
Requirement already satisfied: wrapt_timeout_decorator<2.0.0,>=1.3.12 in c:\users\martinmaursundrishov\appdata\local\packages\pythonsoftwarefoundation.python.3.11_qbz5n2kfra8p0\localcache\local-packages\python311\site-packages (from kintree) (1.5.1)
Requirement already satisfied: pyparsing>=2.4.7 in c:\users\martinmaursundrishov\appdata\local\packages\pythonsoftwarefoundation.python.3.11_qbz5n2kfra8p0\localcache\local-packages\python311\site-packages (from cloudscraper<2.0.0,>=1.2.71->kintree) (3.2.1)
Requirement already satisfied: requests>=2.9.2 in c:\users\martinmaursundrishov\appdata\local\packages\pythonsoftwarefoundation.python.3.11_qbz5n2kfra8p0\localcache\local-packages\python311\site-packages (from cloudscraper<2.0.0,>=1.2.71->kintree) (2.32.3)
Requirement already satisfied: requests-toolbelt>=0.9.1 in c:\users\martinmaursundrishov\appdata\local\packages\pythonsoftwarefoundation.python.3.11_qbz5n2kfra8p0\localcache\local-packages\python311\site-packages (from cloudscraper<2.0.0,>=1.2.71->kintree) (1.0.0)
Requirement already satisfied: retrying>=1.3.3 in c:\users\martinmaursundrishov\appdata\local\packages\pythonsoftwarefoundation.python.3.11_qbz5n2kfra8p0\localcache\local-packages\python311\site-packages (from digikey-api<2.0.0,>=1.0.0->kintree) (1.3.4)
Requirement already satisfied: inflection>=0.3.1 in c:\users\martinmaursundrishov\appdata\local\packages\pythonsoftwarefoundation.python.3.11_qbz5n2kfra8p0\localcache\local-packages\python311\site-packages (from digikey-api<2.0.0,>=1.0.0->kintree) (0.5.1)
Requirement already satisfied: certauth>=1.3.0 in c:\users\martinmaursundrishov\appdata\local\packages\pythonsoftwarefoundation.python.3.11_qbz5n2kfra8p0\localcache\local-packages\python311\site-packages (from digikey-api<2.0.0,>=1.0.0->kintree) (1.3.0)
Requirement already satisfied: urllib3>=1.25.3 in c:\users\martinmaursundrishov\appdata\local\packages\pythonsoftwarefoundation.python.3.11_qbz5n2kfra8p0\localcache\local-packages\python311\site-packages (from digikey-api<2.0.0,>=1.0.0->kintree) (2.3.0)
Requirement already satisfied: cookiecutter<3.0.0,>=2.6.0 in c:\users\martinmaursundrishov\appdata\local\packages\pythonsoftwarefoundation.python.3.11_qbz5n2kfra8p0\localcache\local-packages\python311\site-packages (from flet<0.25.0,>=0.24.1->kintree) (2.6.0)
Requirement already satisfied: fastapi<1,>=0 in c:\users\martinmaursundrishov\appdata\local\packages\pythonsoftwarefoundation.python.3.11_qbz5n2kfra8p0\localcache\local-packages\python311\site-packages (from flet<0.25.0,>=0.24.1->kintree) (0.115.8)
Requirement already satisfied: flet-runtime==0.24.1 in c:\users\martinmaursundrishov\appdata\local\packages\pythonsoftwarefoundation.python.3.11_qbz5n2kfra8p0\localcache\local-packages\python311\site-packages (from flet<0.25.0,>=0.24.1->kintree) (0.24.1)
Requirement already satisfied: packaging<24.0,>=23.1 in c:\users\martinmaursundrishov\appdata\local\packages\pythonsoftwarefoundation.python.3.11_qbz5n2kfra8p0\localcache\local-packages\python311\site-packages (from flet<0.25.0,>=0.24.1->kintree) (23.2)
Requirement already satisfied: qrcode<8.0.0,>=7.4.2 in c:\users\martinmaursundrishov\appdata\local\packages\pythonsoftwarefoundation.python.3.11_qbz5n2kfra8p0\localcache\local-packages\python311\site-packages (from flet<0.25.0,>=0.24.1->kintree) (7.4.2)
Requirement already satisfied: uvicorn<1,>=0 in c:\users\martinmaursundrishov\appdata\local\packages\pythonsoftwarefoundation.python.3.11_qbz5n2kfra8p0\localcache\local-packages\python311\site-packages (from uvicorn[standard]<1,>=0->flet<0.25.0,>=0.24.1->kintree) (0.34.0)
Requirement already satisfied: watchdog<5.0.0,>=4.0.0 in c:\users\martinmaursundrishov\appdata\local\packages\pythonsoftwarefoundation.python.3.11_qbz5n2kfra8p0\localcache\local-packages\python311\site-packages (from flet<0.25.0,>=0.24.1->kintree) (4.0.2)
Requirement already satisfied: flet-core==0.24.1 in c:\users\martinmaursundrishov\appdata\local\packages\pythonsoftwarefoundation.python.3.11_qbz5n2kfra8p0\localcache\local-packages\python311\site-packages (from flet-runtime==0.24.1->flet<0.25.0,>=0.24.1->kintree) (0.24.1)
Requirement already satisfied: httpx<1,>=0 in c:\users\martinmaursundrishov\appdata\local\packages\pythonsoftwarefoundation.python.3.11_qbz5n2kfra8p0\localcache\local-packages\python311\site-packages (from flet-runtime==0.24.1->flet<0.25.0,>=0.24.1->kintree) (0.28.1)
Requirement already satisfied: oauthlib<4.0.0,>=3.2.2 in c:\users\martinmaursundrishov\appdata\local\packages\pythonsoftwarefoundation.python.3.11_qbz5n2kfra8p0\localcache\local-packages\python311\site-packages (from flet-runtime==0.24.1->flet<0.25.0,>=0.24.1->kintree) (3.2.2)
Requirement already satisfied: repath<0.10.0,>=0.9.0 in c:\users\martinmaursundrishov\appdata\local\packages\pythonsoftwarefoundation.python.3.11_qbz5n2kfra8p0\localcache\local-packages\python311\site-packages (from flet-core==0.24.1->flet-runtime==0.24.1->flet<0.25.0,>=0.24.1->kintree) (0.9.0)
Requirement already satisfied: click<9.0,>=8.1 in c:\users\martinmaursundrishov\appdata\local\packages\pythonsoftwarefoundation.python.3.11_qbz5n2kfra8p0\localcache\local-packages\python311\site-packages (from mouser<0.2.0,>=0.1.3->kintree) (8.1.8)
Requirement already satisfied: dill>=0.3.9 in c:\users\martinmaursundrishov\appdata\local\packages\pythonsoftwarefoundation.python.3.11_qbz5n2kfra8p0\localcache\local-packages\python311\site-packages (from multiprocess<0.71.0,>=0.70.12->kintree) (0.3.9)
Requirement already satisfied: decorator>=3.4.0 in c:\users\martinmaursundrishov\appdata\local\packages\pythonsoftwarefoundation.python.3.11_qbz5n2kfra8p0\localcache\local-packages\python311\site-packages (from validators<0.20.0,>=0.19.0->kintree) (5.1.1)
[... cut ...] 0.0,>=2.6.0->flet<0.25.0,>=0.24.1->kintree) (0.1.2)
Requirement already satisfied: pycparser in c:\users\martinmaursundrishov\appdata\local\packages\pythonsoftwarefoundation.python.3.11_qbz5n2kfra8p0\localcache\local-packages\python311\site-packages (from cffi>=1.12->cryptography<45,>=41.0.5->pyopenssl->certauth>=1.3.0->digikey-api<2.0.0,>=1.0.0->kintree) (2.22)`

When typing 'kintree' to run, I get `'kintree' is not recognized as an internal or external command,
operable program or batch file.`

T0jan, 2025-02-25:
hi @martin323232,

looks like your pip doesn't add programs to the PATH system variable, so you need to do that manually, see here how to do that: https://stackoverflow.com/a/23709194

martin323232, 2025-02-26:
Hi @T0jan,

thanks for the tip. I had installed Python from Windows Store. Deleted the Store version and downloaded and installed Python, worked as expected after that.

closed 2025-02-26 by martin323232