sparkmicro/Ki-nTree #278: InvenTree API login from Ki-nTree "...InvenTree server is not connected...

anotherstevest2, 2025-03-27:
Gist:  WSL Docker install of Inventree, Ki-nTree install on the same WSL instance, chrome access to inventree works.  Ki-nTree connection to inventree using the same credentials and the path: http://inventree.localhos fails with:
`Server connection error: <class 'requests.exceptions.ConnectionError'>
InvenTree server is not connected. Skipping authentication check
[TREE]  Error connecting to InvenTree server: invalid address, username or password`

Discussion and more details on reddit and inventree-python repo. See:
https://www.reddit.com/r/InvenTree/comments/1jkc0ma/inventree_api_login_from_kintree_inventree_server/
https://github.com/inventree/inventree-python/issues/233#issuecomment-2755835814 (my comments from 3/27/2025)

T0jan, 2025-04-07:
hi @anotherstevest2 

There are some errors plopping up while using Ki-nTree with docker. Some solutions were posted here but this may not cover your case. What I gather from the other sources it more looks like a problem with the inventree API than anything on our end. I also want to say that you try to use it with the 0.18 development version of inventree which is considered unstable at this moment and not officially supported by Ki-nTree. We always only test against the latest stable version (0.17.10 as of writing this post)

anotherstevest2, 2025-04-09:
(copied from the associated Inventree-python issue)

> At this point, I've punted.  I wish I had the skills in this area to help debug this as I *think* my attempted configuration should be fairly common.  That said, my work-around was to bring Inventree up on a re-purposed laptop which is now running native Ubuntu (still using the docker configuration) and, Ki-nTree (on my original machine) has no trouble connecting to it.  Obviously, I'd rather be running both Ki-nTree and Inventree on the same machine but this approach isn't too painful.

closed 2025-05-19 by eeintech